DrA1ex/ff5m #1: GuppyScreen support?

loss-and-quick, 2025-02-28:
Maybe we should also add `GuppyScreen` as one of the alternatives?

DrA1ex, 2025-03-02:
Currently, there are no plans for this. However, I do plan to further improve the functionality of the Feather screen.

In the future, there’s a possibility of developing an alternative screen platform—something similar to Guppy, but different. Personally, I’m not a fan of Guppy or Klipper screens. I believe the original FlashForge screen is better, but it has its own issues, including high resource usage.

By the way, if you’re interested in the Guppy screen, you can try the Klipper mod

loss-and-quick, 2025-03-02:
Okay, what about buildroot env, without it this mod can not be extended, no difference with zmod

DrA1ex, 2025-03-03:
There is a **Buildroot** environment (you can refer to the **Features** section in the [README](/README.md#features)). It's the same as what is used in Zmod and almost the same as what is used in the Klipper mod.

You can check, for example, [S00init # init_chroot](https://github.com/DrA1ex/ff5m/blob/f8d33949dd2bf2b69195f7878eb25508d8d27345/.shell/S00init#L137) and [S99root](https://github.com/DrA1ex/ff5m/blob/f8d33949dd2bf2b69195f7878eb25508d8d27345/.shell/S99root).

I'm in touch with the Zmod developer, so there isn't really much difference in the feature list between this mod and Zmod (since this mod is a fork). Many features or fixes are discussed together and then implemented in both Zmod and this mod.

The Zmod developer is now adapting GuppyScreen, adding some new features to it, but he refuses to share the source code (this is a LICENSE violation since GuppyScreen is under GPL-3). So the only way to use it is to install Zmod — which is wrong because Guppy is open source project.

Anyway, since Zmod is a closed mod and dev won't collaborate, and because some essential features are missing, I had to fork and develop the mod myself. A lot of features are implemented differently, and many strange firmware behaviors have been fixed (which the Zmod dev refuses to fix since the stock firmware behaves the same way).

This mod was originally meant to be as Zmod+, but it's not compatible with the original Zmod anymore.

I understand that everyone wants as many features as possible — and they want them now, not later — but this mod is still in development and not ready for a public release yet. I have not announced it anywhere so far.

For now, I treat this as more of an experimental project for myself. When it gets closer to public release, I’ll decide on the main direction for the mod. Adding GuppyScreen now would just result in cloning Zmod or the Klipper mod — it’s not something that would interest anyone, in my opinion. I’ve rewritten all the inherited Zmod code, and I’ve made it much better for my taste, but the end user doesn’t really care about the code itself. That’s why I think experimentation is the only way forward for this mod, like developing an alternative screen (e.g., Feather or maybe a new one with a real GUI).

If those experiments don’t lead anywhere, it might be better to port the fixes and macros into the Klipper mod since it already has its own active community and is quite popular.

`ftxui` is an interesting library — I'll consider using it if I get stuck with my own `typer` development. Thanks!

loss-and-quick, 2025-03-03:
Okay, i understand you vision

closed 2025-03-03 by loss-and-quick