DrA1ex/ff5m #2: Normal buildroot environment?

loss-and-quick, 2025-03-03:
As mentioned in #1:

> You can check, for example, [S00init # init_chroot](https://github.com/DrA1ex/ff5m/blob/f8d33949dd2bf2b69195f7878eb25508d8d27345/.shell/S00init#L137) and [S99root](https://github.com/DrA1ex/ff5m/blob/f8d33949dd2bf2b69195f7878eb25508d8d27345/.shell/S99root).

In this [mod](https://github.com/xblax/flashforge_ad5m_klipper_mod), you can see a standard [buildroot](https://github.com/xblax/flashforge_ad5m_klipper_mod/tree/master/build_scripts) environment. It would be great if this mod could be made independent of zmod and properly integrated with the buildroot environment.

DrA1ex, 2025-03-05:
Can you describe what you're missing in the current buildroot setup?
You can still do `chroot` to it if you need to.

As I mentioned earlier, the buildroot environment here is almost identical to the one used in the klipper mod.
The key difference is that the klipper mod runs entirely within that environment, while this mod uses it only for certain services.

This approach has been intentional, as the mod is designed to extend the stock functionality rather than replace it (like the klipper mod does).

DrA1ex, 2025-03-26:
Closed due to lack of details.

For now, to get custom tools:
- Use Klipper’s Buildroot configs (no point cloning to Forge-X)
- Build custom binaries using [these instructions](https://github.com/DrA1ex/ff5m/blob/main/.bin/src/README.md) (includes examples)
- Install directly via Entware