DrA1ex/ff5m #3: Opkg not working

Luk3du, 2025-03-03:
When installing z mod, and following the instructions (used multiple versions) opkg files are in /opt/bin but when trying to ssh opkg code it shows "command not found"

DrA1ex, 2025-03-05:
It looks like there is an issue with the PATH environment variable. Normally, opkg should already be preinstalled in ZMod, but it's possible that either the installation process failed or the PATH wasn't updated correctly.

You can bypass the PATH issue by running the opkg command directly using its full path:
```bash
/opt/bin/opkg install ...
```

If this doesn't work, could you please provide the full log of what actions you performed and the exact output or error messages you received?

DrA1ex, 2025-03-09:
I’ve updated the [Installation Guide](https://github.com/DrA1ex/ff5m/blob/main/docs/INSTALL.md) to address the issue with firmware naming.  
Please try installing the mod using the updated instructions. If it still doesn’t work, feel free to reopen this issue.

closed 2025-03-09 by DrA1ex

DrA1ex, 2025-03-10:
It's not. It was a PATH issue.
Fixed in latest version.

Luk3du, 2025-03-10:
So glad you figured out the issue! Sorry I've been so busy so not had any time to test but will get on it asap :)