DrA1ex/ff5m #4: Do I need to uninstall earlier Klipper mod first? I'm unable to get ZMod confirmation and unable to SSH

DougJoseph, 2025-03-09:
I booted with the ZMod file on the USB, and did not see any confirmation that it completed. 

When I tried to SSH into root, on the chance it had succeeded, the old SSH password from the previous Klipper mod, was still in place, except I was caught in a "password reset then disconnect" loop.

Do I need to uninstall earlier Klipper mod first?  I guess I should just try that.

DougJoseph, 2025-03-09:
Ok. I uninstalled the previous Klipper mod…

[image]

…then powered down, edited the USB to have ZMod file, put the USB back in printer, and powered up. First I see spinning thing with “initialization” …

[image]

…and then I see the stock UI:

[image]

Re. The install docs:

“Successful installation will be indicated on the display when finished.”

I’m not seeing that. 

My printer’s firmware is 3.1.5

DrA1ex, 2025-03-09:
Hi Joseph,

Glad to see you here! A long time ago, I came across your YouTube video about installing the Klipper mod. It's a small world, and such coincidences are always pleasant!

First of all, I think the Klipper mod **can** interact with the booting process. While it is completely detached from the main firmware and runs entirely in its own chroot environment, its boot script may run after this mod's boot script (since they are named `S00init` and `S00klipper_mod`, where `i` comes before `k`), and that can break the booting process of both the Klipper mod and Forge-X mod. I haven't tested this scenario yet and will think of how to avoid this interference. 

However, it should work in double-boot mode: when you need Klipper_mod, you insert a USB with `SKIP_MOD`, and when you need this mod, you use `klipper_mod_skip`. 

You asked about ZMod — it works similarly, except it will operates in dual-boot mode without issues. All you need to do is place a USB with `klipper_mod_skip`, as described in the Klipper mod documentation [Dual Boot](https://github.com/xblax/flashforge_ad5m_klipper_mod/blob/master/docs/INSTALL.md#dual-boot).

Now, about the Zmod installation confirmation issue: I think the ZMod dev changed this behavior a long time ago but haven't updated the manual. ZMod now installs without confirmation — it simply reboots the printer and deletes the firmware image file from the USB (a behavior I find somewhat strange).
After successful installation, your printer should boot with stock firmware but will have SSH access and Moonraker/Fluidd available.

To check whether it's installed, try to access `http://PRINTER_IP/` (without specifying any port, as Fluidd runs on port 80, which is the default).

Accessing the printer via SSH may be tricky after installing the Klipper mod since both mods use different SSH private keys, and your system may not allow connection due to a certificate mismatch. To fix this, you need to remove the printer's IP from `known_hosts`.

In Windows:
You can remove it using Regedit at: `HKEY_CURRENT_USER\Software\SimonTatham\PuTTY\SshHostKeys`.

In Linux/macOS:
You need to edit the `~/.ssh/known_hosts` file and remove the line corresponding to your printer's IP.

After that, you should be able to connect via SSH. ZMod (and Forge-X) use these credentials: `root/root`.

About the SSH password: I think ZMod does nothing with it after installation, so it should remain unchanged if you update it. However, it resets the password during installation. Note that the Klipper mod SSH password will not work here, as it is relevant only to its separate chroot env.

I tried changing the password to check how it work, and it was changed and wasn't reset after reboot:
```bash
# Change password in interaction mode 
passwd

# Or with with pipe
yes your_new_password | passwd
```

DougJoseph, 2025-03-09:
Thanks for the awesome response. 

I’m pretty sure the zmod firmware file was still present on the USB. Based on you saying it should have been deleted, I think it is not getting installed. 

I will do some more checking. 

Thanks again!

DrA1ex, 2025-03-09:
During installation, Zmod creates a `zmod_install.log` file on the USB. You can upload it here so I can help check what really happened.

I am also performing a full install-uninstall check from scratch right now and will update the documentation to make this process smoother.
Currently, there is an installation/uninstallation issue in Forge-X itself, and I will upload a fixed version soon.

DougJoseph, 2025-03-09:
OK, I have checked the USB drive several times after attempted installs, and there has never been either a log file present or the deletion of the firmware file, so somehow it’s not working for me. By contrast, the same USB drive has worked for both install of the previous Klipper mod, and for the uninstall of it. So I don’t think it’s a problem with the USB drive or its formatting.

DrA1ex, 2025-03-09:
I completely forgot to mention in the instructions that Zmod requires you to rename the firmware file to `Adventurer5M-zmod-1.0.4.tgz` for the non-pro printer version.

This is probably the reason why the firmware didn’t even try to install.
Could you please rename the file and try again? That should fix the issue.

DougJoseph, 2025-03-09:
Cool. I will rename the file and try again!

DougJoseph, 2025-03-09:
Ok renaming the Zmod file got me definite progress on the first step! 

Here are pics of what that success looked like!

[image]
[image]
[image]
[image]
[image]
[image]
[image]

DrA1ex, 2025-03-09:
Glad to see the installation completed!

Unfortunately, at this point, the Forge-X mod contains a few issues that I’m working to fix. Nothing critical (the bugs I’ve already identified and fixed), but it would be better if you wait a little while before installing Forge-X itself. I’ll post a message here once everything is ready.

If you can’t wait, you can proceed, but you’ll need to fix the `printer.cfg` config manually. The installation creates a duplicate section that prevents Klipper from running. To resolve this, simply delete the `[led chamber_light]` section from the `printer.cfg` file using Fluidd.

DougJoseph, 2025-03-09:
OK. Got it.

My first order of business was to see if I could login via SSH.

After entering the command 
` SSH root@192.168.1.14`
...I first got the warning message about the fingerprint not matching. So, I opened the "known_hosts" file located at (on my Mac): /Users/dougjoseph/.ssh/known_hosts  — and there I removed (deleted) the line with a fingerprint showing for the IP address of my printer, saved the file, and then I again entered the command
` SSH root@192.168.1.14`

... and entered yes at this this question
`Are you sure you want to continue connecting (yes/no/[fingerprint])?`

...at which point I was prompted for a password, and I attempted to use "root" ...

And finally, the last line shown came after that:
`[1586] Mar 09 18:50:55 wtmp_write: problem writing /dev/null/wtmp: Not a directory`

Here's a screen shot from the terminal session:

[image]

**I'm not sure what to make of that last line.**

DougJoseph, 2025-03-09:
OK, so apparently I am logged into root via SSH. In simply ignoring the warning shown at the end of the last post above, I then attempted to follow the install steps second task:

> Download the patched mjpeg-streamer *.ipk files: [link](https://github.com/DrA1ex/mjpg-streamer/releases)
> Upload the files to the printer and install the dependencies as follows:

The first command (to store my printer's IP address inside a variable) seems to have been accepted:
`PRINTER_IP=192.168.1.14` 

Then on the second command, Im guessing that the "./" before "mjpg-streamer-*" ...
`scp -O ./mjpg-streamer-* root@$PRINTER_IP:/opt/packages/`

... would need to have the file path on my Mac to the folder where I downloaded all the .ipk files. 

So, I tried this version of the line / command:

`scp -O /Users/dougjoseph/Documents/3D/Flashforge/Adventurer\ 5M/Firmware/Klipper\ Mod\ for\ AD5M\ \(newer\ updated\ mod\)/mjpg\ streamer\ files/mjpg-streamer-* root@$PRINTER_IP:/opt/packages/`

...But that got me a response saying
`scp: unrecognized option: O`

So I then tried a version of the line without the -O:

`scp /Users/dougjoseph/Documents/3D/Flashforge/Adventurer\ 5M/Firmware/Klipper\ Mod\ for\ AD5M\ \(newer\ updated\ mod\)/mjpg\ streamer\ files/mjpg-streamer-* root@$PRINTER_IP:/opt/packages/`

...And that got a response of:
`/usr/bin/dbclient: No such file or directory`

**So, I'm not sure what next step is needed here on the uploading of the mjpg-streamer .ipk files.**

Here's a screen shot:

[image]

DrA1ex, 2025-03-09:
It seems like you're viewing an outdated version of the [Installation Instructions](https://github.com/DrA1ex/ff5m/blob/main/docs/INSTALL.md#temporary-run-the-switching-script). I've simplified the installation process — you now only need to run a single command, which will download and install everything automatically.

Regarding the SSH warning: I attempted to investigate this issue, but it appears to be related to something in the bundled binaries. For now, I simply ignore the warning. It's possible that some binaries may need to be rebuilt to eliminate this message.

As for the `scp -O` flag: This is required because the printer’s OS distribution doesn't support the new SFTP protocol. The `-O` flag enables legacy mode in `scp`.  

It's odd that your version of `scp` doesn't support this option. I’ve run the same command on macOS, and my version includes that flag. You might need to update your system software or try installing an alternative version of `openssh` via `brew`.

For reference, I’m running macOS 15.1, but I believe older versions should also support the `-O` flag, since it's specifically for legacy compatibility.

In the updated installation instructions, you no longer need to manually transfer files to the printer. Alternatively, you can upload files via **Fluidd** — just drop them into the G-code list section. Once uploaded, the files will be available in the `/data` folder.

DougJoseph, 2025-03-09:
> It seems like you're viewing an outdated version of the [Installation Instructions](https://github.com/DrA1ex/ff5m/blob/main/docs/INSTALL.md#temporary-run-the-switching-script). I've simplified the [...]

Cool!

I just refreshed the page at https://github.com/DrA1ex/ff5m/blob/main/docs/INSTALL.md

... and I will give it another go!

DougJoseph, 2025-03-09:
OK, I tried the single command line for downloading the resource, and I got this:

```
# CURL=/opt/cloud/curl-*/bin/curl; $CURL -kfsSL https://raw.githubusercontent.com/DrA1ex/ff5m/refs/heads/main/switch.sh | bash
Download requirements...
######################################################################## 100.0%
######################################################################## 100.0%
######################################################################## 100.0%
bash: line 22: opkg: command not found
Failed to install dependencies!
```
Note the last parts:
> **bash: line 22: opkg: command not found**
> **Failed to install dependencies!**

[image]

DougJoseph, 2025-03-09:
For what it's worth, at least I definitely can access Fluidd at http://192.168.1.14  

[image]

DrA1ex, 2025-03-09:
It seems like this is the same issue mentioned in issue #3.  
Initially, I thought it was unrelated to ZMOD and instead linked to an installation problem.  

What’s strange is that I reinstalled the mod around 10 times today and never encountered this issue before.  

First of all, we need to determine if you have the `opkg` binary installed.  
To check this, run the following command with the full path:  
```bash
/opt/bin/opkg update
```  

If this works, you will need to override the `PATH` variable and then execute the `switching.sh` script:  
```bash
export PATH="/bin:/sbin:/usr/bin:/usr/sbin:/opt/bin:/opt/sbin:/opt/bin:/opt/sbin"
CURL=/opt/cloud/curl-*/bin/curl; $CURL -kfsSL https://raw.githubusercontent.com/DrA1ex/ff5m/refs/heads/main/switch.sh | bash
```  

However, it might be better to hold off on this for now. It’s late in my location, so I won’t be able to provide immediate assistance with any potential installation issues that arise.  

For now, you should have a working ZMOD setup, but keep in mind that if any hidden bugs appear after switching, certain functions (e.g., printing) may not work properly.  

I’ll be available tomorrow to continue further!

DougJoseph, 2025-03-09:
OK, here's what I get when I enter `/opt/bin/opkg update`

```
# /opt/bin/opkg update
Downloading http://bin.entware.net/armv7sf-k3.2/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
#  
```
So then I ran the command you told me to run after that... and it looked like this (seemed successful):

```
Updated list of available packages in /opt/var/opkg-lists/entware
# export PATH="/bin:/sbin:/usr/bin:/usr/sbin:/opt/bin:/opt/sbin:/opt/bin:/opt/sbin"
# CURL=/opt/cloud/curl-*/bin/curl; $CURL -kfsSL https://raw.githubusercontent.com/DrA1ex/ff5m/refs/heads/main/switch.sh | bash
Download requirements...
######################################################################## 100.0%
######################################################################## 100.0%
######################################################################## 100.0%
Downloading http://bin.entware.net/armv7sf-k3.2/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
Installing mjpg-streamer (1.0.1-1) to root...
Installing libjpeg-turbo (3.0.3-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libjpeg-turbo_3.0.3-1_armv7-3.2.ipk
Installing libv4l (1.28.0-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libv4l_1.28.0-1_armv7-3.2.ipk
Installing mjpg-streamer-input-uvc (1.0.1-1) to root...
Installing mjpg-streamer-output-http (1.0.1-1) to root...
Configuring libjpeg-turbo.
Configuring libv4l.
Configuring mjpg-streamer.
Configuring mjpg-streamer-input-uvc.
Configuring mjpg-streamer-output-http.
Installing busybox (1.37.0-4) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/busybox_1.37.0-4_armv7-3.2.ipk
Installing htop (3.3.0-2) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/htop_3.3.0-2_armv7-3.2.ipk
Installing nano (8.3-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/nano_8.3-1_armv7-3.2.ipk
Installing liblzma (5.6.2-2) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/liblzma_5.6.2-2_armv7-3.2.ipk
Installing libbz2 (1.0.8-1a) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libbz2_1.0.8-1a_armv7-3.2.ipk
Installing libmagic (5.45-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libmagic_5.45-1_armv7-3.2.ipk
Installing zsh (5.9-3) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/zsh_5.9-3_armv7-3.2.ipk
Installing libcap (2.69-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libcap_2.69-1_armv7-3.2.ipk
Configuring busybox.
Warning: '/opt/bin/find' is already exists, skipping.
Warning: '/opt/bin/grep' is already exists, skipping.
Warning: '/opt/bin/ash' is already exists, skipping.
Warning: '/opt/bin/netstat' is already exists, skipping.
Warning: '/opt/bin/sh' is already exists, skipping.
Warning: '/opt/bin/fgrep' is already exists, skipping.
Warning: '/opt/bin/egrep' is already exists, skipping.
Warning: '/opt/bin/xargs' is already exists, skipping.
Warning: '/opt/sbin/ifconfig' is already exists, skipping.
Warning: '/opt/sbin/route' is already exists, skipping.
Warning: '/opt/sbin/ip' is already exists, skipping.
Configuring liblzma.
Configuring libbz2.
Configuring libmagic.
Configuring nano.
Configuring libcap.
Configuring zsh.
Configuring htop.
Remove old repository data...
Restarting Moonraker...
Waiting moonraker to start...
Switching repository...
ZMOD source changed successfully!

Rebooting...
# Connection to 192.168.1.14 closed by remote host.
Connection to 192.168.1.14 closed.
```

Here's a screen shot:

[image]

And a few seconds after that, I heard the little MIDI tune of success play again!

Your help is so welcomed!

DougJoseph, 2025-03-09:
Re. "I’ll be available tomorrow to continue further!"

THANK YOU!!!! :-)

DougJoseph, 2025-03-10:
No rush, nothing pressing. But whenever you get a chance to take a look....

I updated Moonraker and it seemed to be fine.

I updated Z-mod and it seemed to be fine. The version numbers shown underneath the update button... said:

`v1.1.0-368-g36a3f93d-inferred -> 1.1.1-180`

However, ever since around the same time I updated, Zmod in particular (maybe?) my ability to log into root, got broken.

I get a hot-pink Flashforge logo screen in the terminal window, and then this:

`-sh: exec: : not found`
`Connection to 192.168.1.14 closed.`

Thoughts? 

See screen shot:

[image]

DrA1ex, 2025-03-10:
It seems this is an installation issue. It's strange that I haven't encountered this before, but I’ve been experimenting with different mods for a long time, so my environment might not be completely clean.

First of all, the SSH issue has the same root cause as the issue with `opkg` not working—most likely, the `PATH` isn't set correctly. This might be due to a problem introduced in some version of zmod.

Secondly, the `SET_MOD_PARAM` issue seems to be a bug related to mod switching. It appears that some files were not initialized properly during the process.

I've already fixed these critical bugs, and the fixes are available in the repository. To resolve them, simply perform an OTA update of the mod via Fluidd.

There are additional minor bugs, but they are not critical. I’ll fix them in the next release, as modifying the main branch directly (hot-fixing it) carries the risk of breaking the printer.

Thank you for being so specific and detailed in your descriptions — it really helps stabilize the installation process.

If you encounter any new bugs, feel free to contact me at any time!

DougJoseph, 2025-03-10:
Sounds great, thanks!

DougJoseph, 2025-03-10:
Regarding tweaking the camera, inside the file at `‎/mod_data/camera.conf` there are some settings, where I set them as follows:
```
E_SHARPNESS=255
E_BRIGHTNESS=0
E_CONTRAST=128
E_GAMMA=10
E_GAIN=1
``

...but in that same page it also says this:

```
# Image post-processing settings.
# You can play with settings and choice the best for you in page:
# http://printer_ip:8080/control.htm
```

So, I visited that URL (but with my printer's IP), and I see the controls:

<img width="557" alt="Image" src="https://github.com/user-attachments/assets/7586e408-1462-4868-8771-a2812fd7702d" />

But no display of the camera, so I loaded another browser window at my camera's live stream URL: http://192.168.1.14:8080/?action=stream

<img width="690" alt="Image" src="https://github.com/user-attachments/assets/550e7045-c327-4f15-81fa-69cf743fa139" />

I am getting a much more usable image quality. I am not sure if my limited changes to the original file caused the improvement, or if the changes I made in the `control.htm` page are causing the improvement, and if it is the latter, and since there is no save button, I do wonder if the changes are being saved, and if so, where are they being saved?

DougJoseph, 2025-03-10:
Hey, one other question, for me and others who are looking to migrate from the older Klipper Mod to this one...

What changes if any, are needed to the "Machine start G-code" of the earlier mod?

```
START_PRINT BED_TEMP=[bed_temperature_initial_layer_single] EXTRUDER_TEMP=[nozzle_temperature_initial_layer]
```
And what changes if any, are needed to the "Machine end G-code" of the earlier mod?

`END_PRINT`

Thanks again for such amazing help!

DougJoseph, 2025-03-10:
OK, I decided to try to guess at unknown Camera parameter values based on the `camera.conf` file having "E_" at the start of each, and based on the name of a param in the html control file at `http://printer_ip:8080/control.htm` — I only did this with the ones that had a single-word name.  

Here's my relatively successful edit of the params in the `camera.conf` file:

```
# revised camera params by Doug Joseph - Design8Studio, March 10, 2025:
E_BRIGHTNESS=40		// out of 64 max
E_CONTRAST=25		// out of 100 max
E_SATURATION=42		// out of 100 max
E_HUE=300		    // out of 2000 max
E_GAMMA=100		    // out of 300 max
E_GAIN=1		    // out of 255 max
E_SHARPNESS=7		// out of 7 max
```

And here's a screen shot of what my camera's output now looks like! Much better:

[image]

[image]

DougJoseph, 2025-03-11:
@DrA1ex 
I seem to have messed up. I tried to reinstall the old Klipper Mod alongside the new. The old one did not work. So then I uninstalled it again. However, that somehow resulted in new mod no longer working. Stuck in permanent "initializing" screen. So I cannot get into the new mod, cannot get to the uninstall macro of the new mod to try to redo it, and cannot get the old mod working. I'm kinda stuck. Any advice?

DrA1ex, 2025-03-11:
Sorry for the delay. Our time zones differ by approximately +12 hours, so when it's daytime for you, it's already nighttime here.

For urgent questions, you can reach me via the Telegram group: [https://t.me/+ihE2Ry8kBNkwYzhi](https://t.me/+ihE2Ry8kBNkwYzhi).

I’ve updated the documentation to be more specific regarding the questions you asked. Here’s a summary:

- **`E_*` camera parameters**: These are meant to be set after performing manual tests on the `control` page.
- In the **`START_PRINT` macro**, you need to define **`BED_TEMP`** and **`EXTRUDER_TEMP`** parameters. More information can be found in the [Slicing page](https://github.com/DrA1ex/ff5m/blob/main/docs/SLICING.md).
- If you are printing using the stock screen (not Feather), make sure to include **M190** and **M104** commands; otherwise, the stock screen won’t work properly. This is also explained in the [Slicing page](https://github.com/DrA1ex/ff5m/blob/main/docs/SLICING.md).
- Regarding the **weight exceeded** error: This is a failsafe protection mechanism designed to prevent damage to the printer's bed due to nozzle impact. The weight limit is measured in grams, with a default value of `1200`, which corresponds to about 1.2 kg of impact force — quite significant. Detailed explanations of this feature can be found in both the [Printing page](https://github.com/DrA1ex/ff5m/blob/main/docs/PRINTING.md) and the [Configuration page](https://github.com/DrA1ex/ff5m/blob/main/docs/CONFIGURATION.md).

---

### Details about the **`weight exceeded` error**:  

This error indicates that your printer has hit the bed with significant force while trying to print. Possible causes include:  
- **A weight cell issue**: The weight cell might need calibration, especially if you manually leveled the bed at some point.  
- **Hardware issues**: There may be other hardware problems affecting your printer.  

To temporarily resolve this issue, you can increase the threshold or disable the weight check feature altogether:

```bash
# Increase threshold
SET_MOD_PARAM PARAM="weight_check_max" VALUE=1800

# Disable the feature
SET_MOD_PARAM PARAM="weight_check" VALUE=0
```

However, I strongly recommend investigating and resolving any underlying issues with your printer’s kinematics after making these changes.

---

### About the uninstallation issue:  
Installing Klipper over the Forge-X firmware has not been fully tested and may result in compatibility problems. However, Klipper mods that are already installed should work fine alongside Forge-X. I’ll test this later and address any issues that may arise.

---

### Regarding the frozen initialization screen:  
I suspect this is due to a bug in the newer Flashforge firmware (version 3.1.*). At this stage, the firmware attempts to connect to the network but freezes if it fails to establish a connection. In your specific case, the problem might be different.  

To diagnose, run the following macro:  
```bash
TAR_DEBUG
```
Attach the generated file here for further analysis.  

For now, if the printer freezes during initialization, reboot it by powering it off and back on. This seems to be a flickering issue, as most of the time the firmware works fine—but occasionally it gets stuck. You might consider downgrading the stock firmware to version 2.7.*, as I’ve only encountered this issue in version 3.1.*. Alternatively, switching to the Feather screen eliminates this issue entirely, as the Feather screen is completely detached and does not rely on the problematic firmware.

---

### About mod uninstallation:  
There are multiple ways to temporarily disable or completely remove the mod. Refer to the [Dual Boot section](https://github.com/DrA1ex/ff5m/blob/main/docs/DUAL_BOOT.md) and [Uninstall page](https://github.com/DrA1ex/ff5m/blob/main/docs/UNINSTALL.md) for more information.

---

I believe the mod is working fine in its current version. I’ve been using it on my printer for two months now without any issues—it works great. Most likely, the problems you’re experiencing are related to documentation clarity.  

Anyway, If the mod still doesn’t work, contact me via Telegram, and we’ll troubleshoot your machine together.

DougJoseph, 2025-03-11:
Thanks for all the great clarifications and suggestions on possible workarounds. Regarding the weight issue what is curious is that even though you said the default is 1200g, I get a continual stream of “weight exceeded” warnings and it lists the weight somewhere in the low 700’s. So that’s far below the 1200g that you mentioned.

DrA1ex, 2025-03-11:
The warning you're seeing is simply a warning. The actual failsafe mechanism is only triggered when the specified threshold, such as `1200g` (or whatever value you set), is exceeded.

You can override the warning value in `user.cfg` by adding the following:

```cfg
[temperature_sensor weightValue]
trigger_value: 700
```

If you press the bed, you might notice the sound indicator activating. Moreover, if you're actively printing and the weight does exceed the threshold, the print will immediately stop due to the Klipper emergency handler.

Thanks for pointing this out! I’ll make sure to document it. People often ignore warnings, and I mistakenly assumed the failsafe had already been triggered in your case.

DougJoseph, 2025-03-11:
> The warning you're seeing is simply a alert. The actual failsafe mechanism is only triggered when the specified threshold, such as `1200g` (or whatever value you set), is exceeded.

Although I used the word 'warning' please note that it has been blocking any forward activity, as I mentioned above (see quote pasted below), in which the weight issue blocked me from completing a mesh. In all instances (both before I manually inputted a max weight threshold and after) the weight reflect in the notices was seemingly too low to have actually triggered. If I can ever get back in, I will manually raise the threshold back to 1200g and retry the bed mesh.

Quote from earlier:

> Update: I attempted to do a bed mesh, at 220C and 55C on the bed. 
> 
> It started taking probe measurements, but halted before completion, saying `!! Probing failed due to printer shutdown` and with lots of warnings about `!! Shutdown due to sensor value exceeding the limit`
> 
> Screen shot:
> 
> [image]

DougJoseph, 2025-03-11:
I'm working to try to get back into a functional system of any kind. I had already tried factory stock firmware v 3.1.3-2.2.3. No joy. I'm currently trying factory stock firmware v 2.7.8-2.2.3. It does not look promising, as the spinning thing on the initialization screen has stopped spinning, and each time that happens, nothing else ever appears, no matter how long I leave it (even hours).

Regarding your suggestion here:

> To diagnose, run the following macro:
> 
> ```
> TAR_DEBUG
> ```
> 
> Attach the generated file here for further analysis.

So far, I cannot run any macro's because I cannot ever get back to a working system. I cannot complete a boot.

DougJoseph, 2025-03-11:
I am currently trying with an empty file name `SKIP_MOD_SOFT` on the USB.

DrA1ex, 2025-03-11:
Yes, I noticed that.

The only way to stop the printer from being damaged if something nasty happens is to execute the  `M112` macro, which is the same command you can run in the **Fluidd** toolbar (as an emergency stop). This is essentially what the failsafe feature does — it stops all activity in an emergency.

After that, you need to use the `FIRMWARE_RESTART` command, and then you can continue printing or proceed with whatever you plan to do.

If the failsafe stops the calibration process, there’s no way the weight won’t reach 1200 kilos or whatever value you set. As I mentioned before, if your weight sensor loses calibration (which always happens when you manually level the bed using screws instead of a belt), it may be the reason.

So, about your issue with booting:

When you flash the factory firmware, does the flashing process actually run?  
If it does, that’s a good sign! It means you can restore your printer without needing any additional steps.

If the modification is interacting with the Klipper mod installation, try placing a file named `SKIP_MOD_SOFT` on the USB drive. This should get you back to the stock firmware while still preserving SSH access. Alternatively, you can even try booting into the Klipper mod.

DougJoseph, 2025-03-11:
Thanks so much. Dude, your knowledge on all of this amazing.

Re. 

> if your weight sensor loses calibration (which always happens when you manually level the bed using screws instead of a belt), it may be the reason.

I have not ever manually leveled or messed with screws, but my sensor may have lost calibration somehow.

Re.

> When you flash the factory firmware, does the flashing process actually run?

No, nothing ever completes, and I never get passed a frozen initialization screen. This has now been the case for all the following:

- Attempted stock firmware v 3.1.3-2.2.3-20250107
- Attempted stock firmware v 2.7.8-2.2.3-20241213
- Attempted stock firmware v 2.7.8-2.2.3-20241213 with empty file named `SKIP_MOD_SOFT` on the USB

I am about to try with the empty file only, and no firmware file in the root of the USB.

DrA1ex, 2025-03-11:
Can you describe what exactly happens when your printer gets stuck?

Did you see the mod's screen displaying the boot process log?
If so, what is it printing?

DougJoseph, 2025-03-11:
> Can you describe what exactly happens when your printer gets stuck?

The printer lost the ability to boot up. I've never tried printing anything after first loading the new mod, because I had not yet completed any bed mesh or vibration calibration. 

> Did you see the mod's screen displaying the boot process log?

Since the printer cannot boot, and also cannot let me log into root via SSH, I don't know how to access any logs.

DrA1ex, 2025-03-11:
**What the boot process typically looks like:**

1. **Flashforge (black and white) logo**
2. **Forge-X boot logo** (with purple Flashforge artwork)  
   - Initialization step (ends with the message: *"Initialization step done"*)  
   - Root initialization step (prints: *"Starting services"*, ends with: *"Done"*)  
3. **Flashforge Screen loading screen** (with a white message box and spinning loader)  
4. **Loaded Screen UI**

If the printer is running Klipper, the process is simpler:  
1. **Flashforge (black and white) logo**  
2. **Loaded Klipper screen or Guppy screen**

At which step does your printer stop progressing in its boot process?  
If the Forge-X Boot Logo appears, note the text printed at the bottom of the screen

DougJoseph, 2025-03-11:
> let’s clarify what happened:
> 

I will try to remember. :-)

0. I uninstalled the Klipper Mod (what I am calling the old mod, whereas I think of yours as the new mod). I also refer to the Klipper Mod as the "earlier" or "previous" mod.

> 1. You had the Forge-X mod installed, which was working fine and booting normally.

Correct.

> 2. You installed the Klipper mod, and it booted up successfully as well. (I’m not sure what you mean by "the old mod" — are you referring to a previous version or another mod you had installed earlier?)

I attempted to re-install the Klipper Mod. I *think* its install process seemed normal, but afterward the printer did not ever work right, I think. Lost ability to boot.

> 3. However, the Klipper mod didn’t work as expected. (Could you elaborate on what exactly went wrong? What behavior didn’t work?)

Lost ability to boot.

> 4. You decided to remove the Klipper mod (using a macro, correct?).

I attempted the Klipper Mod's uninstall by placement of the uninstall file on the USB and booting.

> 5. After removing the Klipper mod, the printer rebooted, but instead of displaying the Forge-X mod interface, it only showed the Flashforge logo with a spinner.

This is correct.

> 6. At some point, the spinner stopped spinning, and the printer never progressed past this screen.

Yes.

> 
> Next:
> 
> 7. You attempted to flash a new firmware, but the flashing process didn’t start — there were no other logos, messages, or indications of progress, correct?

Yes. I tried to flash both the stock firmwares recommended on your Uninstall Doc page. Neither one seems to flash right.

> 8. After that, you tried various firmware versions, along with the `SKIP_MOD_SOFT` file, but none of those attempts worked either.

Correct.

> 9. You flashed Zmod firmware, it's worked but then you go again to the non spinning spinner.
> 
> Is this everything you’ve done so far?

I think the above is what happened. I've slept since some of it, and my memory is not super sharp on some of it.

DougJoseph, 2025-03-11:
> I think I figured out why you couldn't flash the factory image. I completely forgot about the distinction between Pro and Non-Pro versions. Can you check the actual file name you try to flash?

In both times of me downloading stock firmware to try to flash it, I was careful to choose the non-pro version when downloading. Here are the file names I downloaded:

Adventurer5M-2.7.8-2.2.3-20241213-Factory.tgz

Adventurer5M-3.1.3-2.2.3-20250107-Factory.tgz

DougJoseph, 2025-03-11:
> You can also try the following combination:
> 
> * An empty file named `klipper_mod_skip`
> * An empty file named `SKIP_MOD_SOFT`
> * A factory image

I'm about to try this!

DougJoseph, 2025-03-11:
I'm trying a flash with both the "Skips" and the earliest recommended stock firmware:

[image]

Alas, that also did not work. Same freezing. I may have corrupt memory and be looking at a bricked mainboard.

Thank you, thank you, for being so kind, so generous, and so helpful.

DrA1ex, 2025-03-11:
I tried to reproduce your issue by switching to an older version from Git to ensure it matches yours.  
Then, I installed the Klipper Mod and uninstalled it using the flash image provided on the Klipper Mod release page. Everything completed without any issues on my side.  

However, I noticed that the firmware process starts without showing the Initialization screen.  
This leads me to believe that the flashing process didn’t even begin when you attempted to flash the factory firmware.  

This theory is further supported by the fact that you **can** still flash the ZMod firmware afterward.  

I don't think your printer is actually bricked since you’re able to flash the ZMod firmware. While this doesn’t fix your printer (as it doesn’t revert the firmware files), it’s still a good sign. So let's try to fix it!

Can you try flashing the Klipper Mod again? Since you were able to do this previously, the issue might be with the factory image itself.  

Also, could you join our Telegram channel using this link? It’ll be much easier to troubleshoot the issue using the messenger:  
[https://t.me/+ihE2Ry8kBNkwYzhi](https://t.me/+ihE2Ry8kBNkwYzhi)

DrA1ex, 2025-03-11:
As your issue has been resolved by flashing back the Klipper Mod image, I am closing this issue. It is no longer related to installation problems.

In the meantime, I will work on creating a restoration image to assist others who might face similar problems in the future. 

There is definitely corruption in the system files.
In the Russian Flashforge community, people report similar issues all the time, even when using the stock firmware.

If you encounter any new issues, feel free to create a new one.

closed 2025-03-11 by DrA1ex