DrA1ex/ff5m #5: Issue regarding FF camera. My custom values in the camera.conf are ignored.

DougJoseph, 2025-03-20:
Issue regarding camera. My custom values in the camera.conf file, are ignored on each boot. In order to get camera output right again, I have to visit the control page every time ( at http://192.168.1.14:8080/control.htm )

DougJoseph, 2025-03-20:
Here are my edits to camera.conf for FF camera:
E_BRIGHTNESS=40		// out of 64 max
E_CONTRAST=25		// out of 100 max
E_SATURATION=42		// out of 100 max		
E_HUE=300		// out of 2000 max
E_GAMMA=100		// out of 300 max
E_GAIN=1		// out of 255 max
E_SHARPNESS=7		// out of 7 max

DougJoseph, 2025-03-20:
Hmm. Update:

I changed each param value line to remove the "comment" afterward, and now the values seem to be loading.

i.e. changed this:

`# E_BRIGHTNESS=40		// out of 64 max`

to this:

`# E_BRIGHTNESS=40'

I'm considering this one solved.

closed 2025-03-20 by DougJoseph

DrA1ex, 2025-03-24:
@DougJoseph, I've rewritten the entire effects system. It will be in the 1.2.2 release.
When the update's ready, could you test it and tell me if anything's still unclear?

DrA1ex, 2025-03-24:
Maybe I used too loud of words :D. Only affects camera config - other parts stay the same

DougJoseph, 2025-03-24:
Got it.