DrA1ex/ff5m #6: No camera

Orgullonacional, 2025-03-23:
Camera fail. cannot see camera in Fluidd.
I do can in the web navigator.

DrA1ex, 2025-03-23:
Please read this page carefully: [Camera Documentation](https://github.com/DrA1ex/ff5m/blob/main/docs/CAMERA.md).
If you still can’t get it to work, let me know what you did, what configuration you used (both in camera.conf and Fluidd), and what you expected to happen.

Matt-Armfield, 2025-03-23:
For me, I couldn't use the stock display with the stock camera.  Once I disabled the stock camera (see the above linked Camera Documentation). and enabled the mod's camera using the command: SET_MOD_PARAM PARAM="camera" VALUE=1. It fixed my issues.

DrA1ex, 2025-03-24:
It should work with both the stock and mod cameras.
Configuration might be a bit tricky, but when done right, everything should just work.

DrA1ex, 2025-03-26:
Fixed in [1.2.2](https://github.com/DrA1ex/ff5m/releases/tag/1.2.2). If anything's not working right, feel free to reopen.

closed 2025-03-26 by DrA1ex