DrA1ex/ff5m #7: !! pin PA8 used multiple times in config

makenb, 2025-03-23:
I just installed this mod and klippy refused to boot, the console said:
!! pin PA8 used multiple times in config
I looked in printer.cfg and PA8 was listed as the heater pin and the LED pin so I commented out the LED section. 
Now it boots and the LED is on but I cant control it...

makenb, 2025-03-23:
Never mind, I went through the auto config and it seems to have resolved itself.

DrA1ex, 2025-03-24:
Ok, feel free to reopen the issue if this ever happens again.

evilknivel, 2025-05-13:
Sadly this is happen to me today as well. Whats the counter measure?

DrA1ex, 2025-05-13:
A simple reboot should likely resolve this issue.

To help investigate further:
- Run the `TAR_DEBUG` macro
- Upload the generated archive here

evilknivel, 2025-05-13:
@DrA1ex i am sorry you are absolutely right. Restart fixed it :-|

DrA1ex, 2025-05-14:
Okay! I'm not entirely sure what happened, so let's keep this open for now.

This seems quite rare, and I suspect it might be a hardware glitch. If that's the case, it should remain infrequent.

If anyone encounters this again:
- Reboot the system
- Run the `TAR_DEBUG` macro (logs persist after reboot)
- Upload the archive here

DrA1ex, 2025-09-01:
@fazadaoc This issue is usually resolved by a simple reboot.
Can you provide logs? See the comment: https://github.com/DrA1ex/ff5m/issues/7#issuecomment-2879486879

adamkingdotnet, 2026-09-20:
Hit this again on 2026-09-19 (AD5M non-Pro, fresh 1.4.1 install over stock 5.1.x) and worked out the root cause — adding it here since the issue was closed without one:

**Root cause:** init is a two-step LED rename. `.cfg/data.init.cfg` + `.cfg/init.cfg` inject `[led chamber_light]` (`white_pin: PA8`) into `printer.cfg`, while `.cfg/init.base.cfg` (`-[led chamber_led]`) is supposed to strip the old `[led chamber_led]` (also `PA8`) from `printer.base.cfg`. If the first step lands and the second doesn't, the merged config contains both sections on the same pin → Klipper halts at boot with `pin PA8 used multiple times in config` → stock screen shows E0120.

That also explains why a reboot fixes it for most people: the next S00init pass completes the migration. On mine the state had settled so every boot re-injected the conflict and reboots never cleared it.

**The same pipeline bites display switching:** `SET_MOD PARAM="display" VALUE="FEATHER"` — and `zdisplay.sh` over SSH, which additionally needs `/usr/sbin` in PATH or `chroot` is "command not found" — regenerate the live configs through this machinery. A half-applied run left no `[led]` section at all → Klipper dies with `Existing command 'SET_LED' not found in gcode_macro rename` (macros still reference `SET_LED`).

**Fix that worked here:** delete `[led chamber_light]` from `.cfg/init.cfg` and `.cfg/data.init.cfg` on the printer (the base `chamber_led` already provides the LED), and clean any copy out of `printer.cfg`. Boots reliably since. Note the `1.4.2-beta-3` tag still ships `[led chamber_light]` in both templates, so this remains reachable on fresh installs.

DrA1ex, 2026-09-20:
Just remove the entire [led chamber_led] section from printer.cfg in Fluidd/Mainsail if it’s there, and reboot.

If it’s not there, just reboot — everything should be fixed automatically.

The reason is the Stock screen, which holds the configuration and reverts it after initialization, right before Klipper starts. This only happens with the Stock screen.

The upcoming 1.4.2 uses Feather as the default screen, so this issue won’t be there anymore.

As for the duplicate LED entry in printer.cfg, that’s most likely from user intervention, probably from accidentally copying printer.base.cfg into printer.cfg. So far, nobody has confirmed that the Stock firmware itself migrates this section there.

Still, the fix is very small, so I added a cleanup for it in 1.4.2 just in case.

DrA1ex, 2026-09-20:
I checked the `firmwareExe` binary from stock firmware, and it confirms the original assumption.

On the first boot after a version change, the stock UI restores `printer.base.cfg` from its embedded copy after Forge-X has already updated the config. That brings the old `chamber_led` on `PA8` back right before Klipper starts.

It also updates `/root/version`, so this restore does not happen on the next cold boot, which explains why the issue fixes itself after a reboot.

Teityan, 2026-10-15:
## Basic Information
Printer Model	Adventurer 5M Pro
Firmware Version	5.1.9
Mod Version	1.4.2-beta-4
Screen Type	STOCK

[SCAN-RESULT.json](https://github.com/user-attachments/files/33473926/SCAN-RESULT.json)
[FORM-NOTES-ja.txt](https://github.com/user-attachments/files/33473927/FORM-NOTES-ja.txt)
[REDACTION-MANIFEST.json](https://github.com/user-attachments/files/33473928/REDACTION-MANIFEST.json)
[TAR_DEBUG-redacted-20261015.tar.gz](https://github.com/user-attachments/files/33473929/TAR_DEBUG-redacted-20261015.tar.gz)

## What did you do?
Klipper failed to start with:

pins.error: pin PA8 used multiple times in config

printer.cfg defines [led chamber_light] on PA8, while printer.base.cfg defines [led chamber_led] on the same pin.

The logs confirm that Forge-X removes the old definition during initialization, but firmwareExe later restores printer.base.cfg from its embedded resources.

Relevant stock firmware log:

recoverCfgFile cfg file invalid, trying recovery

recovery from resource success

This reintroduces the duplicate pin assignment before Klipper starts.

## Additional Information

I implemented a local workaround that keeps the stock chamber_led as the only LED object and redirects Forge-X LED commands to it.

Verification results:

Klipper reached Ready.

Two consecutive Klipper RESTART commands succeeded.

A full power cycle also succeeded.

During that power cycle, firmwareExe restored printer.base.cfg again, but the PA8 conflict did not return.

STOCK screen mode and Moonraker connectivity were preserved.

The workaround modifies four initialization template files and macros/base.cfg.

This appears related to #7. I can provide the tested configuration diff and logs.

Could the configuration handling be adjusted upstream to prevent this conflict when using the STOCK screen?

DrA1ex, 2026-10-15:
We can't simply switch back to `chamber_led`. Forge-X intentionally uses `chamber_light` to distinguish its own LED commands from those sent by the stock firmware. This allows Forge-X to prevent the stock screen from overriding its LED control when needed.

I think it makes more sense to investigate why `firmwareExe` considers `printer.base.cfg` invalid and restores it in the first place. Ideally, we need a way to prevent this unnecessary recovery.

DrA1ex, 2026-10-15:
The stock firmware checks `/root/version`, and if it doesn't match, it resets `printer.base.cfg` and `printer.cfg`, right after Forge-X has already updated them. That's how it works in 5.1.1.

I also tested the latest 5.1.9 and found that it checks not only the version but also the file size. Since Forge-X modifies these files, their size may fall below the threshold, causing the stock firmware to reset them on every boot.

It's a bit of a hacky approach, but it's actually meant to migrate configuration changes between firmware versions (for example, the servo angles have changed).

So I've updated Forge-X to handle this migration itself, before the stock firmware gets a chance to reset the configs. This fix is already included in the latest 1.4.1/1.4.2. Hopefully, it fixes the issue for good.

closed 2026-10-15 by DrA1ex