DrA1ex/ff5m #8: HOMING

alund1990, 2025-03-24:
If my Y axis is in its home position already when i got to home all axis, it doesn't move it out and back to home. Instead it just keeps trying to drive the Y axis to the back of the printer and cause some not so nice noises.

DrA1ex, 2025-03-24:
Can you provide more details?

After you turned on the printer:
- Did you home the axes, or was this its first homing?
- Which axes were homed before you started this homing process?
- Did you move the toolhead before starting homing?
- Did you use G28 macro or the Fluidd button to home?

DrA1ex, 2025-03-26:
Fixed in [1.2.2](https://github.com/DrA1ex/ff5m/releases/tag/1.2.2). If anything's not working right, feel free to reopen.

closed 2025-03-26 by DrA1ex