DrA1ex/ff5m #12: 5M Pro Connection and camera issues after removing mod.

Profesyor, 2025-03-27:
Uninstalled MOD, connection issues. Even after uninstalling, and going back to OEM firmware 3.1.5, printer keeps, going offline, idle, offline, idle. Camera also doesn't work. Reflashing firmware, also didn't help. Also tried Firmware Adventurer5MPro-3.1.3-2.2.3-20250107-Factory, my connection issues persisted and camera still doesn't work in Orca-Flashfore on Flash Maker.

DrA1ex, 2025-03-27:
Upload here any logs left on your USB disk after you flash.

What version of the mod did you flash?
Why did you uninstall the mod, was something not working right?

Also, describe more about what you did and what didn’t work.

It’ll help to figure out the reason and how to fix it. The more you describe, the faster we can sort it out

DrA1ex, 2025-03-27:
And when you say about 'disconnecting', you mean losing Wi-Fi connection?

The mod does nothing with Wi-Fi or Ethernet modules, as it's part of the kernel. Along with OrcaSlicer/FlashPrint - as it's part of Screen module (stock firmware actually installs only itself alone, not touching any other part of system).

This is further confirmed because even when using Klipper mod it still doesn't work. Klipper mod uses a dedicated system, meaning it's not related to firmware or core system. If something doesn't work there, it's most probably hardware (not 100% though).

This also relates to the camera. Klipper has its own dedicated streamer, just like Stock and ForgeX. They all implement it as an extension — not replacing anything in the system (as the kernel doesn’t provide anything like that).

And about the USB - which will work and which won't - it's 100% related to the kernel module, which is static.
The mod itself may not recognize some types of filesystems (in 1.2.1), but in 1.2.2 it's fixed.
And in any case, images flashed by original firmware, so it decides whether to or not.

For your network problems: try connecting via Ethernet cable if you're using Wi-Fi - or try Wi-Fi if you're on Ethernet.
For the camera: Try detaching and reattaching the cable;  Check the screen connection too.

If that doesn't work, flash this image: [Adventurer5MPro-debug.tgz](https://github.com/user-attachments/files/19486603/Adventurer5MPro-debug.tgz). It collects logs onto your USB drive (debug.tar) - upload it here then.

DrA1ex, 2025-03-27:
I see you mentioned viewing the camera stream in a browser.
If you're on Stock, where were you trying to view the stream then?

stevelenti, 2025-03-27:
My camera had issues going back to stock as well.  The fix for me was to reset the printer to factory through the menu.  Run the [Adventurer5MPro-ForgeX-uninstall.tgz](https://github.com/DrA1ex/ff5m/releases/download/1.2.0/Adventurer5MPro-ForgeX-uninstall.tgz) then run [3.1.3 Factory images](https://github.com/ghzserg/zmod/tree/main/%D0%A0%D0%BE%D0%B4%D0%BD%D0%B0%D1%8F_%D0%BF%D1%80%D0%BE%D1%88%D0%B8%D0%B2%D0%BA%D0%B0)

At first I had just ran the uninstall macro REMOVE_MOD and the camera would not work on stock firmware.

-Steve

DrA1ex, 2025-03-27:
Regarding the camera issue, it’s highly unlikely to be caused by the mod itself. In nearly all cases, this occurs when the documentation isn’t followed closely.

Many users, including those with the Pro version’s built-in camera, have reported successful operation.
Personally, I’ve been using a custom camera with this setup for three months without software related problems

stevelenti, 2025-03-27:
When going from the Klipper Mod to Zmod, I did not follow the Zmod install instructions correctly.  I completely skipped the steps of resetting the printer to factory and flashing the factory image before flashing Zmod.  So you could be correct that it was not following the documentation correctly which eventually led me to have issues with your mod.  I did get everything fixed both on your mod and the stock firmware.  Running 1.2.2 with no issues now.

-Steve

DrA1ex, 2025-03-28:
Issue resolved by flashing Uninstaller + Recovery.
Problem was most likely caused by installing the mod through unpacking the firmware image to USB

closed 2025-03-28 by DrA1ex