DrA1ex/ff5m #13: LED ON / LED OFF Macros

Matt-Armfield, 2025-03-28:
These macros are reversed for me.  I hit LED OFF and the LEDs go full strength and vice versa with LED ON.

AndyAD5M, 2025-03-28:
Hey,I can't tell you what the problem is in your case, but the easiest way for me would be to rename the macro alias "led off" to "led on" and vice versa, or to look in the mod/base.cfg and possibly edit it.

[image]
[image]

DrA1ex, 2025-03-29:
> These macros are reversed for me. I hit LED OFF and the LEDs go full strength and vice versa with LED ON.

How did you connect your LED? Is it a custom one?
If your LED has inverted control (like mine for example), try the 'invert' parameter.

DrA1ex, 2025-03-30:
Related to the documentation, changed it in 1.3.0