DrA1ex/ff5m #14: E0011 host error after successful install

draskostar, 2025-03-28:
I was on 3.1.5 and successfully updated to 1.2.2. When loaded the firmware there some strange buzzer appeared. When I tried to go home host error appeared. I had to uninstall to stock and printer works again. I tried same with 2.4.5 and still host error.

DrA1ex, 2025-03-29:
@draskostar, Unfortunately, E011 errors mean everything. Could you flash this image (rename it to match your printer model): [Adventurer5MPro-debug.tgz](https://github.com/user-attachments/files/19486603/Adventurer5MPro-debug.tgz). It collects logs onto your USB drive (debug.tar) - upload it here then

draskostar, 2025-03-29:
Hello,

thank you for fast response. I can do it. Meanwhile I tried also zmod from Sergei and it works without any problems. I am on flashforge 5m pro. I tried also 1.3.0 and same error. 

[image]

draskostar, 2025-03-29:
[debug.zip](https://github.com/user-attachments/files/19522725/debug.zip)

DrA1ex, 2025-03-29:
You need to calibrate your bed’s weight sensors:
https://github.com/DrA1ex/ff5m/blob/main/docs/FAQ.md#why-am-i-getting-a-weight-exceeded-error

draskostar, 2025-03-29:
I do not know why all my *.cfg files are empty. No edit is possible:(

draskostar, 2025-03-29:
> You need to calibrate your bed’s weight sensors: https://github.com/DrA1ex/ff5m/blob/main/docs/FAQ.md#why-am-i-getting-a-weight-exceeded-error

I did it. Thank you very much.

closed 2025-03-29 by draskostar

phillipmoak, 2025-07-09:
i just finished installing 1.3.4 without having any prior firmware mods. the update was successful, and everything booted up like it was supposed to but i got this same error message. the printer seems to be functioning properly so im not sure why i have this error? should i ignore it?