DrA1ex/ff5m #15: [Issue]: Installation error

WatWall, 2025-03-30:
### Printer Model

Adventurer 5M

### Firmware Version

3.1.3

### Mod Version

1.3.0

### Other Mods Installed

- [ ] I have installed Klipper Mod earlier.
- [ ] I have installed ZMod earlier.
- [ ] I have installed ForgeX earlier.

### Screen Type

- [x] I use Stock screen.
- [ ] I use Feather screen.

### What did you do?

I installed the mod but on the fluidd screen I see this error and printer doesn't function.
```
Option 'gcode' in section 'gcode_macro UPDATE_MCU' must be specified

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
```

### What didn't work?

```
Option 'gcode' in section 'gcode_macro UPDATE_MCU' must be specified

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
```

### Checklist

- [x] I have read the documentation and did not find an answer.
- [x] I have read the F.A.Q.
- [x] I have tried rebooting the printer, and the problem persists.

DrA1ex, 2025-03-30:
You checked that you not installed Zmod earlier, but this macro used in it. It's not standard macro and it not used in ForgeX.  
So, it either you used another mod earlier, or you probably edited configuration files, adding some macro to them, and did it wrongly You checked that you not installed Zmod earlier, but this macro used in it. It's not standard macro and it not used in ForgeX.  
So, it either you used another mod earlier, or you probably edited configuration files, adding some macro to them, and did it wrongly.
Since you didn’t upload any debug information I can’t say more.

WatWall, 2025-03-30:
Sorry, it was my fault. I used zmod before and reverted to the stock firmware. Thought I deleted all data that comes with zmod but seems I couldn't. Found the errored macro and fixed the issue.

closed 2025-03-30 by WatWall