DrA1ex/ff5m #16: [Issue]: When I upload to print, it gets stuck in standy and never starts the print

DocJ-GA, 2025-03-31:
### Printer Model

Adventurer 5M

### Firmware Version

2.6.5

### Mod Version

1.3.0

### Other Mods Installed

- [ ] I have installed Klipper Mod earlier.
- [ ] I have installed ZMod earlier.
- [ ] I have installed ForgeX earlier.

### Screen Type

- [x] I use Stock screen.
- [ ] I use Feather screen.

### What did you do?

I sliced a file in orca slicer and uploaded to printer.  It shows:
`Running Command {zsend}...:
Response: CMD M23 Received.
File opened: Right - Back.gcode Size: 7844504
File selected
ok

Command {zsend} finished
`

then goes into standby and never starts the print.

Gcode start is as described in slicing.md

### What didn't work?

Printing

GCODE OF START PRINT
`; MD5:602823439863d80f756141239fbdacc8  
; HEADER_BLOCK_START
; generated by OrcaSlicer 2.2.0 on 2025-03-30 at 20:20:45
; total layer number: 24
; filament_density: 1.23
; filament_diameter: 1.75
; max_z_height: 10.11
; HEADER_BLOCK_END

; THUMBNAIL_BLOCK_START

;
; thumbnail begin 140x110 2240
; iVBORw0KGgoAAAANSUhEUgAAAIwAAABuCAYAAAD1TPu3AAAGV0lEQVR4Ae2d6XEbORCFp6xbpcO6S9
; RF3bfkEBSCQ1AIzsAhKASFoBAcgkNQCAyBi+ctbPWOOeQQA8w0gPdVvdpfu0V7mo3X8xrcoiCEEEII
; IYTkxdLS0rf5+fl+Qcgker3e2/Pz8/Dh4WGws7PzoyBkFOgqV1dXv1EsUufn579mZma+FoRY0EnKhS
; KFbrO+vv69IHmDztHv9z/GFYsUjit2m0xZWVl5QeeoWyxWOLZwfBUkH6yxdZU1xOw2iYPOcHt7+9mk
; WMqGmON3okwytk26zebm5mtB0gDHBjoBHu7T09MwRNFAR0dH7zyiIgfffFsoJycnQzMRDc2RFKxoYI
; g5fkcKvvH2QaJIDg4OhsfHx8PLy8tgBUNDHCFVxvbu7m54f38ftFhoiCNjb2/vZ1sFMU23KYguqnIg
; LWIepYhQ43KIbkND3CHT5kBaxDyqA3AEueRAWsQ8qkWa5kBaREMcGO3G1lUcvwMQi7Ft0m1oiD0Qq7
; F1FQ1xA2wOlJtoiB2QOVCOYh5VE5sDIV2+ublpNfvRKBriMcgcCKsIpsv8+SdCwzYfUsidGdduwwUt
; wahxGd0FuyuhVxHKhWIL9eLiQkWxSHFBq9A1LqNgsC+Dgjk7O1PxmcrKdkEL3xSNxhZF8/j4qOozlZ
; WdIfa9uZ+rYIiTH79TyYG0KNk8KtUcSIuSWtBKPQfSoujzqNxyIC2KMo9yvehO+VFUeRSNrQ6pH785
; LuuUiVzeC23Q2OoTXkKenp4Ot7e3fxVakBfdKT1CYHt4ePjnerDpMDpG7vJFd8iYraj+YlMUglMUCm
; SfR+cmWOZA2FlBYNfGRXeqWvY5oFDw5S3vEnVigquMLQK73JeduhQ6ie0qk1YzWlvK0nbRnfrXDmBv
; CIUCz1L3t2+CLmUxB9Kp6+vr/7oKisZlU9D7UhbHZZ3CglfZ2LrKy1IWcyCdKhtbc7R4+e82MsTMgX
; QKE2hdY+sqp6Us0+4G2jbncxa6CrqJNbahb1FMvZRl3qcM8Eq57esd1N+S43Lbi+m1l7JQMHDdEJw4
; u037wt+5NLZ4Dl18jlpLWbJgIFNp3swVNVmhjG0TYWWlsmBMuvkuC8b+IDIul3X9wVOXzIG0xCy2gM
; d2GfPq+JvsNDhL8YeBtN/biVHS2I7KgbqSnMyKSczOzvbRbWCA0RYhFA8NsV9NkwO1JTxrW8D7+/uD
; hYWFl6Iu5r3MqymSAf4w8qiiIW4mu+A0bQ4UWjJy2NjYcNvWwxG1u7v7IQuGhthdcsHJNQcKITmZTd
; VVqjAJ51u5aPAtgSFmt6mnUQtOXUtOZljpNO9f+oUv0G3Mt+OzXDg0xPUfipZxGZIFbOxHuGu35fHb
; dhstZ7EmtZEDTSs5maGrzM3NhV/hXF5e/l5+0WfHcB5R/5822siB6kpOZuZN7lvRJuMMcc7jt5w2tP
; xAkZzMph6XfQNDXO42uRpiDTlQWXIyU3NXaZwhzmH8jsHYGhvxWmijyhCnnEdpzoFaNbauLC4uvowy
; xKmN31pzoE6NrSs4oqq6TQqGWGMOJK+edG5sXVlbW/sxqtvEmkeVcyAtxlZOZqou4btQZYhjy6PktK
; HJ2HrPgbQQcx6VXQ6khfKClnZDPOmie1eSkUPQHEgLMeRRGo2tnMxgbFWPy77Rmke5XnRvs4CdF5xi
; R1se5eOie4gCtpNZcsbWFQ2G2OdFd1/CEaQuB9JCVwtavnIgGFH8++iOPopcfQ6khTbzKDltNM2B5P
; 9NrkmBd7LgFDtVhhjfOh/fXvlQ8JB9jMv4XMYDDZoUS5Q5kBbKeRR+fBHtHl6jiSHu8qL7uGJTs+AU
; OzaPslECJhmXPKo8bWjJgTC2J5MDaQHdxhwdn/AxtmCmyaPktKE1B6KxDcC48bvqoTAHypy6eVQMF9
; 1XV1d/FiQ8X758+Touj4ohB6Kx7YCq8Zs5EKmkKo/SkgMlu+AUO9IQa1jKorGNAJlHdbmU1dpFd+IH
; a4jbXspiDhQx0hC3sZTFHCgBpCEOtZTFHChB7A8G+F7KUnnRnfjBtyHmglMmSEPsspQV1UV34gf5gw
; HTLGVJY8scKDPkgtakHwlI4qI78YP8wYBRS1ny6snW1tZHQQi6Ta/X+11eymIORMZi8ygcUdghZg5E
; JmIXtPB+hTkQqQ27CiGEEEIIIRnyD+jeEhNZ9FqIAAAAAElFTkSuQmCC
; thumbnail end
; THUMBNAIL_BLOCK_END

; external perimeters extrusion width = 0.62mm
; perimeters extrusion width = 0.62mm
; infill extrusion width = 0.62mm
; solid infill extrusion width = 0.62mm
; top infill extrusion width = 0.60mm
; support material extrusion width = 0.60mm
; first layer extrusion width = 0.70mm

; EXECUTABLE_BLOCK_START
M73 P0 R122
M106 S0
M106 P2 S0
;TYPE:Custom
START_PRINT EXTRUDER_TEMP=220 BED_TEMP=70
M190 S70
M104 S220
G90
G21
M83 ; use relative distances for extrusion
`

### Debugging Information

[debug_20250331_002815.tar.gz](https://github.com/user-attachments/files/19528220/debug_20250331_002815.tar.gz)

### Checklist

- [x] I have read the documentation and did not find an answer.
- [x] I have read the F.A.Q.
- [x] I have tried rebooting the printer, and the problem persists.

DrA1ex, 2025-03-31:
Are you sure to read this?
https://github.com/DrA1ex/ff5m/blob/main/docs/FAQ.md#stock-screen-freezes-i-cant-print-anything

This also should be fixed just by rebooting your printer

DocJ-GA, 2025-03-31:
Update.  I checked that, did the correct save_config and did do a power cycle twice to no effect before posting as I had already read that.  The printer screen wasn't freezing but going unresponsive for a few seconds at a time.  I reconnected the ribbon cables and it still was having the problem.  The only thing I had changed was added spoolman to the config.  I removed the mod and reinstalled, added spoolman and now all things are working correctly again.  I'm not sure what was happening.

closed 2025-03-31 by DocJ-GA