DrA1ex/ff5m #17: [Issue]: "Weight Exceeded" and "Bed pressure detected" Issues

jwidess, 2025-04-06:
### Printer Model

Adventurer 5M

### Firmware Version

3.1.5

### Mod Version

1.3.1

### Other Mods Installed

- [ ] I have installed Klipper Mod earlier.
- [ ] I have installed ZMod earlier.
- [ ] I have installed ForgeX earlier.

### Screen Type

- [x] I use Stock screen.
- [x] I use Feather screen.

### What did you do?

After installing the mod and attempting a `AUTO_FULL_BED_LEVEL` it started but then resulted in a "Weight Exceeded" error. Followed the instructions [here](https://docs.google.com/document/d/1Oou4A56g5HTrxBAMoH-bTnTZZ3IZyGr_3jL9tUYYiow/edit?usp=drivesdk) to calibrate and after that disabled the stock screen for Feather screen and then tried bed levelling again which resulted in the error below. 
```javascript
// Shutdown due to sensor value exceeding the limit
// Once the underlying issue is corrected, use the
// "FIRMWARE_RESTART" command to reset the firmware, reload the
// config, and restart the host software.
// Printer is shutdown
``` 
Then after `FIRMWARE_RESTART` I keep seeing these in the Console, `!! Bed pressure detected: 1060.0 g.` errors. Now I'm unsure if I should re-enable the stock screen and do the calibration again, or if something else is wrong.

### What didn't work?

Primary problem is "!! Bed pressure detected: 1060.0 g." errors and prior "Weight Exceeded" errors which go away after calibrating, but come back after a reboot.

### Debugging Information

[debug_20250406_041148.tar.gz](https://github.com/user-attachments/files/19619612/debug_20250406_041148.tar.gz)

### Checklist

- [x] I have read the documentation and did not find an answer.
- [x] I have read the F.A.Q.
- [x] I have tried rebooting the printer, and the problem persists.

jwidess, 2025-04-06:
**Update**: The errors stopped appearing for a few hours, and I was able to successfully complete a print, however after running `AUTO_FULL_BED_LEVEL` again the errors are reappearing.

DrA1ex, 2025-04-06:
Everything I can explain is documented here: [FAQ: Weight Exceeded Error](https://github.com/DrA1ex/ff5m/blob/main/docs/FAQ.md#why-am-i-getting-a-weight-exceeded-error)

This is a hardware/calibration issue - unrelated to the mod.
If you can't get it work, just disable the feature.

I specialize in the mod's software - for hardware issues, you'll need to handle that yourself.

jwidess, 2025-04-06:
> All available explanations are documented here: [FAQ: Weight Exceeded Error](https://github.com/DrA1ex/ff5m/blob/main/docs/FAQ.md#why-am-i-getting-a-weight-exceeded-error).
> 
> This issue is related to hardware or calibration, not the mod itself. If you’re unable to resolve it, disabling the feature is always an option.
> 
> My expertise focuses on the mod’s software. For hardware-related questions, you may want to consult other users who have experience with similar issues

@DrA1ex As I explained, I've ran the calibration via the stock screen which resolved the issue temporarily. However now that I am using Feather screen I can't perform the calibration again, and I doubt it will fix it permanently if I do. My question to you is as this seems to be a software calibration issue, is there a way to perform this calibration or set the bed weight `H1` offset via your FORGE-X software? And if this isn't a software problem, do you have any idea what hardware problem could cause this issue? I haven't found any information regarding this online.

DrA1ex, 2025-04-06:
Bed meshing temporarily resolved the issue because it's called inside `LOAD_CELL_TARE`, which performs taring (resets weight to zero). However, this is not a permanent fix - after reboot, you'll get whatever raw value your weight sensor returns.

I've explained the core issue here: https://t.me/FF_ForgeX/1179

This could be either:
- A factory defect
- Sensor degradation over time

I don't have solutions for this issue. If probing completely stops working, Flashforge support will likely send replacement components.

My only recommendation is to:
1. Skip the mod at boot using the `SKIP_MOD` macro
2. Recalibrate using Flashforge's official instructions
3. Try to perform recalibration with the bed heated to about 60°C

jwidess, 2025-04-06:
@DrA1ex Thanks for the help! I have confirmed that something seems to be wrong with my beds weight sensor as I can do stock calibration at 60C with a 500g weight reading as ~480g however when the bed cools to 25C it will show 700-1000g however this value is negative as applying pressure to the bed lowers the value. I will contact Flashforge support to see what they advise I do. In the meantime I will use `LOAD_CELL_TARE` in my bed levelling after the bed is heated before a print to hopefully circumvent this issue.

DrA1ex, 2025-04-06:
No manual taring needed - it's automatically called in all relevant macros.

If you've already recalibrated and the weight isn't too high, but you still get warnings: adjust the `weight_check` limits (see F.A.Q for details).

jwidess, 2025-04-09:
@DrA1ex Well I went ahead and disassembled the bed, inspected the 4x pressure sensors and the control board, everything looked fine so I re-assembled, and it seemed to be fine for a day or two, but today a few hours into a print the print was aborted due to "sensor value exceeding the limit" again. I'll be disabling `weight_check` with  `SET_MOD_PARAM PARAM="weight_check" VALUE=0` for the time being. I'll keep an eye on the WeightValue "temperature" sensor to see if I can notice a pattern in the fluctuations. Thanks again for your development on this software!

DrA1ex, 2025-04-10:
@jwidess You likely can't fix these sensors without replacing them if they're showing such inconsistent values.
On my printer, temperature changes only cause about 300-400g of variation.

As a workaround, you could adjust the weight limit to better match your specific range (the maximum supported value is around 2048g)