DrA1ex/ff5m #18: [Issue]: PAUSE macro resumes print after parking if filament_switch_sensor is enabled

vlad-iakovlev, 2025-04-06:
### Printer Model

Adventurer 5M

### Firmware Version

2.7.8

### Mod Version

1.3.1

### Other Mods Installed

- [ ] I have installed Klipper Mod earlier.
- [ ] I have installed ZMod earlier.
- [x] I have installed ForgeX earlier.

### Screen Type

- [ ] I use Stock screen.
- [x] I use Feather screen.

### What did you do?

1. On mod v1.3.0 printed gcode with `PAUSE` command in the middle to change filament manually.
2. Printed just fine, pause command worked as expected and I was able to change filament using "LOAD FILAMENT" macros in Fluidd.
3. Noticed a new version of mod (v1.3.1) and updated to it. Also updated Fluidd to the latest version.
4. Started print again from the same gcode stored on printer.

### What didn't work?

While printing after update, `PAUSE` was executed, printer parked, made sound and then instantly (or may be after a second) continued printing like I have pressed "Resume", but I didn't.

### Installation/Uninstallation Logs (if applicable)

Unfortunately I rebooted printer after the issue so I don't have log saved. I'll try printing something small with the `PAUSE` in it and will share it's gcode and logs later today.

### Checklist

- [x] I have read the documentation and did not find an answer.
- [x] I have read the F.A.Q.
- [x] I have tried rebooting the printer, and the problem persists.

### Additional Information

After described in `What didn't work?` happened I stopped print, waited for printed to cool down and turned it off with the switch on the back. The next day I tried printing the same gcode (again, the same one which was stored on the printer) one more time and the same happed.

Then I checked diff on Github and checked `filament_switch_sensor` value by executing `GET_MOD_PARAM PARAM=filament_switch_sensor`. I noticed that it enabled and then disabled it: `SET_MOD_PARAM PARAM=filament_switch_sensor VALUE=0`, confirmed that it disabled. Restarted the print and this time it worked as on v1.3.0 - print head parked and printer waited until I did what I needed to do.

Please see gcode attached if you need it. I'm ready to help with debugging this issue.

[Screw Tray 30mm.gcode.txt](https://github.com/user-attachments/files/19621948/Screw.Tray.30mm.gcode.txt)

vlad-iakovlev, 2025-04-06:
And now it works as expected. Idk, I guess update didn't set `filament_switch_sensor` parameter properly and something checks for it to be 0 and something to be 1.

I'll reopen issue later if I notice it again.

Btw, thank you for the mod, I really love having proper web ui to control it remotely and not being forced to use resistive touch screen in 2025.

closed 2025-04-06 by vlad-iakovlev

DrA1ex, 2025-04-06:
Thank you for the detailed explanation of the issue.
I'll run some tests on my printer too.
If I discover any problems, I'll report back here