DrA1ex/ff5m #21: [Feature]: Power loss recovery

skylatian, 2025-04-20:
### Printer Model

Both

### Feature Description

Apologies if this already exists! I searched the code and it doesn’t seem like
there’s anything there. Power loss recovery would be fantastic, but I’m not sure if FF has done something tricky to get it to work. Would that be possible at all in this mod?

### Purpose of the Feature

Would bring this firmware closer to feature parity with stock FF firmware

### Other Mods Installed

- [ ] I have installed Klipper Mod earlier.
- [ ] I have installed ZMod earlier.
- [ ] I have installed ForgeX earlier.

### Screen Type

- [ ] I use Stock screen.
- [ ] I use Feather screen.

Matt-Armfield, 2025-04-28:
Forge-X works with the stock firmware so if you run the stock screen and lose power, you have the ability to recover the print.  This option does not appear on the feather screen since that screen only shows minimal and custom information.

pmduque, 2025-05-14:
I'm really not sure how to upvote this feature, but I'm really interested in it.

DrA1ex, 2025-08-14:
I released a preview version of **Power Loss Recovery** for Feather/Guppy screens.
It should work fine overall, but there are limitations in this preview version.

**What doesn't work now:**
- Z-Offset won't be restored
- Fan state won't be restored
- Velocity limits won't be restored
- Pressure advance setting won't be restored
- Skew profile won't be restored
- Nozzle won't be cleaned/prepared (but this matches stock behavior)
- If your last print used KAMP, the default bed mesh will be loaded

**Workaround:** You can set these parameters yourself before restoring the print, so I consider it better to have this version than nothing.

**How to enable the feature:**
1. Update to the latest OTA version
2. Enable the option:
```
SET_MOD PARAM='power_loss_recovery' VALUE=1
```

**How to recover a print:**
After reboot, do:
- `RESURRECT` to restore the print
- `RESURRECT_ABORT` to abort

Or click `Restore` in the dialog window on Fluidd/Mainsail or Guppy screen.

DrA1ex, 2025-09-07:
Released in version 1.4.0

closed 2025-09-07 by DrA1ex