DrA1ex/ff5m #25: [Issue]: cam not connecting to Octoeverywhere

NT1216, 2025-05-07:
### Printer Model

Adventurer 5M

### Firmware Version

3.1.5

### Mod Version

1.3.3

### Other Mods Installed

- [x] I have installed Klipper Mod earlier.
- [ ] I have installed ZMod earlier.
- [ ] I have installed ForgeX earlier.

### Screen Type

- [x] I use Stock screen.
- [ ] I use Feather screen.

### What did you do?

Installed ForgeX.  Have everything working and can see Cam in Fluidd interface on web UI and in orcaslicer.

### What didn't work?

Get the message "Unable to load webcam" on Octoeverywhere dashboard.   All other features show...bed and extruder temp etc.       Cam works in Fluidd Web UI and Orcaslicer.   Using cam in ForgeX not factory cam settings. 

### Debugging Information

na

### Checklist

- [x] I have read the [documentation](https://github.com/DrA1ex/ff5m/tree/main?tab=readme-ov-file#documentation) and did not find an answer.
- [x] I have read the [F.A.Q](https://github.com/DrA1ex/ff5m/blob/main/docs/FAQ.md).
- [x] I have tried rebooting the printer, and the problem persists.

DrA1ex, 2025-05-08:
Try asking in the Discord community or OctoPrint-specific communities.
This issue is not related to the mod, as it works in Fluidd. The printer exposes the stream at:
http://your_IP:8080/?action=stream

There is no firewall, so everything running on the printer should be accessible on the local network.
If you need to access it from an external network, you will need to find a way to tunnel through a remote server to make it available on the internet.

NT1216, 2025-05-08:
I reinstalled the companion app for octoeverywhere and it resolved the issue.

closed 2025-05-08 by NT1216