DrA1ex/ff5m #26: [Issue]: Fluidd Not Loading in Slicer - Causing printer to freeze completely

Sterling1998, 2025-05-12:
### Printer Model

Adventurer 5M

### Firmware Version

3.1.5

### Mod Version

1.3.3

### Other Mods Installed

- [x] I have installed Klipper Mod earlier.
- [ ] I have installed ZMod earlier.
- [x] I have installed ForgeX earlier.

### Screen Type

- [x] I use Stock screen.
- [ ] I use Feather screen.

### What did you do?

Printer froze about 2.5 hours into a print - after rebooting the printer I went to go into fluidd in Orca Slicer to pull a log. No error codes.

### What didn't work?

Printer froze about 2.5 hours into a print - after rebooting the printer I went to go into fluidd in Orca Slicer to pull a log. Now I get a white screen or a view of my other printer in Orca Slicer. I am able to get a view of the printer in Fluidd via my browser.

[debug_20250512_181835.tar.gz](https://github.com/user-attachments/files/20172478/debug_20250512_181835.tar.gz)

### Debugging Information

[debug_20250512_181835.tar.gz](https://github.com/user-attachments/files/20172478/debug_20250512_181835.tar.gz)

### Installation/Uninstallation Logs (if applicable)

NA

### Checklist

- [x] I have read the [documentation](https://github.com/DrA1ex/ff5m/tree/main?tab=readme-ov-file#documentation) and did not find an answer.
- [x] I have read the [F.A.Q](https://github.com/DrA1ex/ff5m/blob/main/docs/FAQ.md).
- [x] I have tried rebooting the printer, and the problem persists.

### Additional Information

I have rebooted the printer several times, rebooted orca several times, rebooted my computer once and the inability to access Fluidd through orca persists. Problem with printer freezing is tied to orca/fluidd somehow as when I tried to access fluid through orca the printer froze.

Sterling1998, 2025-05-12:
Thank you for your reply. 

When I check my IP address I have it typed in correctly, I know this because orca has the test button and it tells me that "Connection to OctoPrint is working correctly" but there is a problem somewhere. When I go to click the device tab it just gives me a white screen. it only does this on the A5M printer, not my Neptune 4 Plus or Qidi Q1 Pro. Before this problem started I was able to connect to the Fluidd interface through Orca. this suddenly stopped working a few hours into having the firmware flashed.

I will check the ram usage while it is printing. The freezing of the printer happens in the middle of a print, not during any commands or anything. Do I need to send the SAVE_CONFIG command.

I don't have a camera attached to the printer.

DrA1ex, 2025-05-12:
You don't need to run SAVE_CONFIG, unless you change something — but Fluidd will prompt you if needed.

There are two addresses: one for sending G-code, and another for the Printer tab. Check the second one.

It should be either: `<IP>` without port number or `http://<IP>/fluidd` 

[image]

Sterling1998, 2025-05-12:
28 minutes in the print froze again - Fluidd will not load. Printer needs rebooted.

DrA1ex, 2025-05-13:
> Now with that fixed, I am still unable to connect to Fluidd through orca

Copy-paste your address from the Orca configuration exactly as it appears. You might have mistyped it without noticing—especially if you previously used Klipper Mod, where the address format is different.

> 28 minutes in the print froze again

Did you slice the G-code using a nearly default profile? Adding extra G-code per layer might freeze the stock screen.
Also was anything displayed on the screen at that moment? Were there any error messages or warnings?

Upload your G-code here, and I'll test it on my printer.

Sterling1998, 2025-05-15:
I ended up going back to stock firmware to confirm I was not having a hardware issue, the printer functions perfect on stock. I have reinstalled Forge-X fresh, following [this](https://www.youtube.com/watch?v=NKDHkpEclkU&ab_channel=ZeroDotCMD) youtube video, and I am at the point where I am starting my first print. I have gotten the host error E0011 twice now, though. Both were tied to the calibration process though. Once before when I started and once right after it finished the calibration process. I have completed the benchy boat and am now doing another print to see. I am able to connect to the printer through orca, I removed the HTTP part under device UI. I have it checking the md5 file, and have added it to the post-processing scripts. After numerous prints today I have not had a failure that was the cause of the firmware. It seems that my earlier issues were tied to a bad install of some kind.

closed 2025-05-15 by DrA1ex