DrA1ex/ff5m #27: mainsail update manager

buenso, 2025-05-13:
### Printer Model

Adventurer 5M

### Feature Description

Would it be possible to add : 

[update_manager mainsail]
type: web
channel: stable
repo: mainsail-crew/mainsail
path: ~/mainsail

In Moonraker.conf 

### Purpose of the Feature

I think that we may be some old geezers left, who still prefer Mainsail over Fluidd 
And if Fluidd is updated, then why shouldn´t Mainsail :-) 

Thanks in advance

### Other Mods Installed

- [x] I have installed Klipper Mod earlier.
- [ ] I have installed ZMod earlier.
- [ ] I have installed ForgeX earlier.

### Screen Type

- [x] I use Stock screen.
- [ ] I use Feather screen.

buenso, 2025-05-13:
Can we safely edit moonrake.conf ourselves, if adding: 

[authorization]
force_logins: true  # Require login for all users

In moddata/user ? 

Don´t want to do anything , if it messes things up.

Thanks

DrA1ex, 2025-05-13:
Forge-X includes an adapted version of Mainsail that resolves known issues (you may have noticed occasional malfunctions in Klipper Mod). Previously, OTA updates weren't possible, but since the Mainsail team continues maintaining the project, I've created a dedicated fork. From now on, Mainsail will support OTA updates in Forge-X.

Thank you for your suggestion!

DrA1ex, 2025-05-13:
My bad - there was a typo in the script - that's why you're seeing the warning.

Use Fluidd to do an OTA update for Forge-X, and then update Mainsail again.
This time you'll also notice the warning about unmatched detected repo, but this time ignore this (as your repo now corrupted and won't match the new one).

closed 2025-05-13 by DrA1ex