DrA1ex/ff5m #30: [Issue]: Some Camera.conf param values don't seem to work or be reflected in control.htm

DougJoseph, 2025-05-21:
### Printer Model

Adventurer 5M

### Firmware Version

3.1.5

### Mod Version

1.3.4

### Other Mods Installed

- [ ] I have installed Klipper Mod earlier.
- [ ] I have installed ZMod earlier.
- [x] I have installed ForgeX earlier.

### Screen Type

- [ ] I use Stock screen.
- [ ] I use Feather screen.

### What did you do?

I've used both Feather screen and Guppy screen. Currently running Guppy screen.

Regarding these two params:

```
E_EXPOSURE_ABSOLUTE=1000
E_EXPOSURE_AUTO="3"

```

...I've been trying for a long time to get "Exposure, Auto" to be set to "Manual Mode" and "Exposure (Absolute)" to be some lower value, 1000 or 500 or so.

I've tried setting the "auto" one to 

`E_EXPOSURE_AUTO="2"`

and to 

`E_EXPOSURE_AUTO="1"`

and to 

`E_EXPOSURE_AUTO="0"`

and to all the above with no quotes:

`E_EXPOSURE_AUTO=2`

and to 

`E_EXPOSURE_AUTO=1`

and to 

`E_EXPOSURE_AUTO=0`

And have tried single quotes:

`E_EXPOSURE_AUTO='2'`

and to 

`E_EXPOSURE_AUTO='1'`

and to 

`E_EXPOSURE_AUTO='0'`

And have tried double quotes with words

`E_EXPOSURE_AUTO="Manual Mode"`

And the same here with single quotes:

`E_EXPOSURE_AUTO='Manual Mode'`

And always saving and rebooting on each try, and reloading the control.htm file each and every time, and I cannot seem to budge the "Exposure, Auto" displayed in control.htm to change. 

PS: Below is a copy and paste of the contents of my camera.conf

```
# Resolution width
WIDTH=640

# Resolution height
HEIGHT=480

# Frame per second
FPS=15

# Video device: 'auto' or video<N> (like video0)
VIDEO=auto

# Reduce camera memory usage
# This may be handy if your camera consumes too much memory.
# For example, even for 640x480 resolution, it may uses memory as it 1080p stream.
# Disable it if you expericencing issues with your camera
REDUCE_MEMORY=1

# Image post-processing settings.
# Enable this to use image post-processing
POST_PROCESSING=1

# You can adjust these settings to fine-tune the camera's image quality.
# Visit the settings page at:
# http://printer_ip:8080/control.htm
# After testing, apply the desired values to the configuration fields below.
# Don’t forget to uncomment the parameters by removing the '#'.

# New as of 2025-05-20
# http://192.168.1.14:8080/control.htm
E_BRIGHTNESS=0
E_CONTRAST=28
E_SATURATION=22		
E_HUE=100
E_WHITE_BALANCE_TEMPERATURE_AUTO=0
E_GAMMA=100
E_GAIN=3
E_POWER_LINE_FREQUENCY=0
E_WHITE_BALANCE_TEMPERATURE=4600
E_SHARPNESS=7
E_BACKLIGHT_COMPENSATION=0
E_EXPOSURE_AUTO_PRIORITY=0
E_JPEG_QUALITY=70
E_EXPOSURE_ABSOLUTE=1000
E_EXPOSURE_AUTO="1"
```

### What didn't work?

I cannot seem to budge the "Exposure, Auto" displayed in control.htm to change. I was trying to get to Manual Mode. 

### Debugging Information

[debug_20250521_034713.tar.gz](https://github.com/user-attachments/files/20359732/debug_20250521_034713.tar.gz)

### Checklist

- [x] I have read the [documentation](https://github.com/DrA1ex/ff5m/tree/main?tab=readme-ov-file#documentation) and did not find an answer.
- [x] I have read the [F.A.Q](https://github.com/DrA1ex/ff5m/blob/main/docs/FAQ.md).
- [x] I have tried rebooting the printer, and the problem persists.

DrA1ex, 2025-05-21:
Exposure modes are sequential (0, 1, 2, 3) based on their order in the pop-up.

[image]

For my camera, Manual mode is 1:

```
E_EXPOSURE_AUTO=1
```

What you see in control.htm doesn't always match your camera's actual settings. These parameters are basically hardware hacks, not something the manufacturer expects you to change.

All these budget cameras use copy-pasted firmware some underqualified developer made years ago, then they just keep reusing it across different models. Unless you have a premium brand like Logitech.

So judge by the actual camera image, not what the UI shows.

DougJoseph, 2025-05-21:
Got it! I’m asking too much from the el cheapo camera! :-) Thanks for all you do!

closed 2025-05-21 by DougJoseph