DrA1ex/ff5m #41: [Issue]: Recurring communication loss every few hours with the Ethernet port

silmaril464, 2025-07-01:
### Printer Model

Adventurer 5M Pro

### Firmware Version

3.1.5-2.2.3

### Mod Version

1.3.4-42-g809d3d22

### Other Mods Installed

- [ ] I have installed Klipper Mod earlier.
- [ ] I have installed ZMod earlier.
- [ ] I have installed ForgeX earlier.

### Screen Type

- [x] I use Stock screen.
- [ ] I use Feather screen.

### What did you do?

Installed Forge-X (everything correct) and normal use of the printer.

### What didn't work?

Hello, Let me start by saying that everything is working great overall, but I’m having some issues with the Ethernet port I only use the Ethernet port for the printer’s LAN connection, but the issue is that every few hours the port either loses connection or gets stuck. I have to manually restore the connection by disabling and re-enabling the Ethernet port from the printer's menu, or by unplugging and reconnecting the Ethernet cable so the network comes back up.

Is no one else experiencing this issue?

### Debugging Information

A few days ago, I contacted Flashforge technical support to check if it was a firmware issue, and they told me to reinstall the firmware just in case, as they had no record of any problems with the Ethernet port...

The thing is, I don’t remember having any issues with the Ethernet port before installing Forge-X, but I’m not completely sure either.

Does no one know anything about this? Is no one else experiencing this? Is this a problem only I’m having, or is it a known issue? I’ve searched but haven’t found any information about it

### Checklist

- [x] I have read the [documentation](https://github.com/DrA1ex/ff5m/tree/main?tab=readme-ov-file#documentation) and did not find an answer.
- [x] I have read the [F.A.Q](https://github.com/DrA1ex/ff5m/blob/main/docs/FAQ.md).
- [x] I have tried rebooting the printer, and the problem persists.

DrA1ex, 2025-07-01:
With the Stock screen, the network is fully controlled by the stock firmware - so your issue is hardware-related. 
The problem could be with: the printer itself, your router, the network cable, etc.

DrA1ex, 2025-07-03:
I'm referring to the Stock firmware to confirm this is almost certainly a hardware issue rather than software. When you switch to Feather (Forge-X), the mod relies on Linux's native Ethernet tools and kernel networking stack - meaning Linux itself handles the connection. If you encounter problems, the first step is to meticulously check your physical hardware: router, cables, and potentially the Ethernet module itself if other components test fine.

I recommend consulting someone knowledgeable or asking in Linux/networking communities, since the printer uses standard Linux networking.

As an alternative, you could try Wi-Fi if Ethernet behave unreliable.