DrA1ex/ff5m #42: [Issue]: KAMP is not working, it is doing a full bed mesh calibration instead

doliveira1987, 2025-07-02:
### Printer Model

Adventurer 5M

### Firmware Version

3.1.5

### Mod Version

1.3.4-7:

### Other Mods Installed

- [ ] I have installed Klipper Mod earlier.
- [ ] I have installed ZMod earlier.
- [ ] I have installed ForgeX earlier.

### Screen Type

- [ ] I use Stock screen.
- [x] I use Feather screen.

### What did you do?

I updated the Forge-X mode to 1.3.4-42-g809d3d22, Fluidd to v1.34.3, and Mainsail to v2.14.1

### What didn't work?

Since the update that includes this "1.3.4-7: Remove override for BED_MESH_CALIBRATE, as it may discourage if use_kamp is enabled" the KAMP is not working, my starting G CODE is this

KAMP_DEFINE_AREA MIN={first_layer_print_min[0]},{first_layer_print_min[1]} MAX={first_layer_print_max[0]},{first_layer_print_max[1]}
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]  FORCE_KAMP=1
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]

The the use KAMP is enabled

Use KAMP: YES
 --> SET_MOD PARAM="use_kamp" VALUE=1

### Debugging Information

NA

### Checklist

- [x] I have read the [documentation](https://github.com/DrA1ex/ff5m/tree/main?tab=readme-ov-file#documentation) and did not find an answer.
- [x] I have read the [F.A.Q](https://github.com/DrA1ex/ff5m/blob/main/docs/FAQ.md).
- [x] I have tried rebooting the printer, and the problem persists.

DrA1ex, 2025-07-03:
I can't reproduce your issue. While I don't use FORCE_KAMP, I do have the use_kamp parameter enabled.
Here's my console output as proof of working KAMP:

```
16:33:34  info Extruder temperature: 250.0
16:33:34  info Bed temperature: 70.0
16:33:34  // State: MD5 CHECK
16:33:34  info Launching MD5 checksum verification
16:33:34  INFO: Begin MD5 check...
16:33:35  INFO: MD5 checksum correct!
16:33:35  info Skipping Z-offset setup, SKIP_ZOFFSET=1
16:33:35  // State: HOMING...
16:33:50  // Initiating KAMP bed leveling
16:33:50  info Started stock nozzle cleaning 250.0/70.0
16:33:50  // State: PREPARE CLEANING...
16:34:00  // State: HEATING...
16:34:00  // Wait bed temperature to reach 70 Cº
```

To debug:
- Upload any G-code file that reproduces the issue
- Reprint to confirm the problem occurs again
- Run `TAR_DEBUG` without rebooting
- Share the generated archive here

CHaucke89, 2025-07-03:
Double check that you have "Label objects" enabled in your slicer, as well. I forgot to enable it in one of my slicer profiles which led to it performing a full bed mesh rather than KAMP when I switched to it.

(Also, fairly certain the M190 and M104 commands in your starting gcode are unnecessary since you're using Feather screen.)

doliveira1987, 2025-07-03:
Thank you, CHaucke89, I forgot to check the "Exclude objects" actually, and now it is working fine.

By the way, thanks for the tip about the M190 and M104, I removed them, and you were right, they are unnecessary.