DrA1ex/ff5m #44: [Issue]: OrcaSlicer Cannot Connect

stevenwalton, 2025-07-17:
### Checklist

- [x] I have searched for [similar issues](https://github.com/DrA1ex/ff5m/issues?q=is%3Aissue) but haven't found any matching reports.
- [x] I have read the [documentation](https://github.com/DrA1ex/ff5m/tree/main?tab=readme-ov-file#documentation) and did not find an answer.
- [x] I have read the [F.A.Q](https://github.com/DrA1ex/ff5m/blob/main/docs/FAQ.md).
- [x] I have tried rebooting the printer, and the problem persists.

### Printer Model

Adventurer 5M

### Firmware Version

FF5M 3.1.5 / 1.3.4-main-47d44b9 @ 17.05.2025 14:28:37

### Mod Version

Forge-X 1.3.4

### Other Mods Installed

- [ ] I have installed Klipper Mod earlier.
- [ ] I have installed ZMod earlier.
- [ ] I have installed ForgeX earlier.

### Screen Type

- [x] I use Stock screen.
- [x] I use Feather screen.

### What did you do?

Added Feather display option 

### What didn't work?

Upload and print from Orca Slicer

### Debugging Information

.

### Additional Information

I recognize there are guides, but I'm not sure they all are appropriate to an issue like this. 

I've been unable to upload prints via OrcaSlicer, but it is giving the message "Could not connect to Flashforge: Connection refused" despite being able to see and control the printer form the Device tab.

I seem to be able to resolve this issue by uninstalling (via blank file on USB) and then reinstalling. But I noticed I hit the issue again after setting the display to `FEATHER`. Specifically, it seems to happen after doing a power cycle (turning off via the switch on the back).

What's interesting is when I'm using the feather display I notice a certain program isn't running...

```bash
kunos :: ~ » ss -lntp
State            Recv-Q            Send-Q                        Local Address:Port                        Peer Address:Port
LISTEN           0                 128                                 0.0.0.0:7125                             0.0.0.0:*                users:(("python3",pid=1713,fd=16))
LISTEN           0                 1000                                0.0.0.0:22                               0.0.0.0:*                users:(("dropbear",pid=1662,fd=3))
LISTEN           0                 9                                         *:80                                     *:*                users:(("httpd",pid=1757,fd=3))
LISTEN           0                 1000                                   [::]:22                                  [::]:*                users:(("dropbear",pid=1662,fd=4))
```

I messed up and didn't grab the logs... but on the boot screen there was an issue with an error about the MCU and something not connecting. When doing the uninstall I noticed something about a gcode command script. Let me try to get a print done and then I'll see if I can reproduce and grab those logs. (Is there a way to get the boot logs when doing the remove? It isn't populated onto the USB like the install script does. If not, I can record with my phone)

To compare, if I immediately log in after doing a removal and install I get this
```bash
kunos :: ~ » ss -lntp
State Recv-Q Send-Q  Local Address:Port Peer Address:Port
LISTEN0      128           0.0.0.0:7125      0.0.0.0:*    users:(("python3",pid=1529,fd=14))
LISTEN0      1000          0.0.0.0:22        0.0.0.0:*    users:(("dropbear",pid=1382,fd=3))
LISTEN0      5       192.168.1.132:8898      0.0.0.0:*    users:(("firmwareExe",pid=1618,fd=34))
LISTEN0      10            0.0.0.0:8899      0.0.0.0:*    users:(("udhcpc",pid=1709,fd=19),("wpa_cli",pid=1682,fd=19),("wpa_supplicant",pid=1680,fd=19),("firmwareExe",pid=1618,fd=19))
LISTEN0      9                   *:80              *:*    users:(("httpd",pid=1568,fd=3))
LISTEN0      1000             [::]:22           [::]:*    users:(("dropbear",pid=1382,fd=4))
```

As we can see here, our expected service is running on 8899. With this, we no longer get the connection refused. 

I'm unsure, but this may be related to #38. Though I am using WiFi and this is the same for both states.

```bash
kunos :: ~ » cat /etc/network/interfaces
# interface file auto-generated by buildroot

auto lo
iface lo inet loopback
```

stevenwalton, 2025-07-17:
I think you're right. I swear it worked at first, which is why I wasn't looking there, plus I had this issue before, which is why I was trying Forge-X (and liking it so far). I'll give it a shot tomorrow.

Idk if your Orca is different but those are not the options I have. I'll assume it is `Octo/Klipper`, or need the manual version. May help others avoid a similar mistake?

[image]

closed 2025-07-17 by stevenwalton

DrA1ex, 2025-07-17:
You need OctoPrint/Klipper. That’s handled by the Moonraker server, but the naming differs across slicers.

stevenwalton, 2025-07-17:
Yeah, got that. I was just mentioning because the doc explicitly says OrcaSlicer