DrA1ex/ff5m #46: [Feature]: Co-Print KCM Integration

Knapp87, 2025-07-22:
### Printer Model

Adventurer 5M

### Feature Description

I'd like a feature that makes my printer prepare me a sandwich in the morning! But more realistically can the Co-Print KCM be integrated into Forge-X? Co-Print points users to the old Klipper Mod and when I tired to use the KCM with Forge-X I was running into issues that I'm not sure how to resolve. Looking at the Co-Print discord it would appear the only way anyone has gotten the Co-Print to work is with the old Klipper Mod even though I know people would rather use Forge-X. 

### Purpose of the Feature

To allow users with the Co-Print KCM to use the superior Forge-X mod.  

### Other Mods Installed

- [ ] I have installed Klipper Mod earlier.
- [ ] I have installed ZMod earlier.
- [x] I have installed ForgeX earlier.

### Screen Type

- [x] I use Stock screen.
- [ ] I use Feather screen.

### Additional Information

Co-Print Discord: https://discord.gg/Qum27Gmh

SMF1208, 2025-07-22:
funny you posted this I was thinking about trying this right now.  I have mine working perfectly with the old mod.  What issues were you running into?

rusvasul, 2025-08-15:
I have some experience with playing with the co print mods both with the xblax mod and this one and I was not able to get it to work with this DrA1ex mod. I have since gone back to my stock print head due to hardware issues with the hotend but once I get those figured out I may swap back and then I would love to see this happen. so in the mean time, Feel free to reach out to me and I will try to remember to check my messages. I have a lot of notes on how I feel about the co print files and hardware. I also just tried out the ad5x for a few weeks but decided to return it because they decided to hide all of the ifs code in the main runtime binaray file and you have no access to those settings which I would argue are incredibly wistful of time and filament.

SMF1208, 2025-08-15:
> I have some experience with playing with the co print mods both with the xblax mod and this one and I was not able to get it to work with this DrA1ex mod. I have since gone back to my stock print [...]

Ironically ive since wrote some documentation and created some hardware mods.  I have a few posts up in their discord and also have a set of custom config files i posted there if interested.

My issue with forgex was due to firmware on the coprint equipment.  Possibly an older version of forgex would work or you could flas custom firmware to the coprint equipment