DrA1ex/ff5m #56: [Feature]: Initial cleaning

AlexUnderKH, 2025-09-18:
### Printer Model

Adventurer 5M

### Firmware Version

3.1.4

### Mod Version

1.4.0

### Feature Description

Please add or change the 2nd initial cleaning option at the start of printing to this:
G90
M83
G1 E-0.2 F800
G1 X110 Y-110 Z5 F6000
G1 Z0.2 F1200
G1 E2 F800
G1 X20 E9 F1000
G1 X-60 E12.5 F1000
G1 E-0.5 F800
G92 E0

### Purpose of the Feature

It's convenient, doesn't leave snot on the nozzle, but lays it neatly on the table

### Other Mods Installed

- [ ] I have installed Klipper Mod earlier.
- [ ] I have installed ZMod earlier.
- [ ] I have installed ForgeX earlier.

### Screen Type

- [x] I use Stock screen.
- [ ] I use Feather screen.

DrA1ex, 2025-09-18:
It's the same code, except for the last retract: -0.5 instead of -0.2.  
I’ve never experienced any issues with this retract length. Why do you think 0.5 would be better?

AlexUnderKH, 2025-09-18:
Oh. Sorry. Hm... I'll check the code. And later add video with starting cleaning

DrA1ex, 2025-09-18:
Your video doesn't show any difference. As I said before, the current implementation is no different from what you suggested, except for the very last command, which is the retraction.

In your first video, that bit of filament is stuck outside the nozzle itself, so no retraction can make it fall.
In your second video, that bit of filament fell in the right corner, but you suggested changing the command in the left corner.

It just doesn't make any sense.

You can override the macro, though, by copying and pasting it into `user.cfg` and making any changes you like.